STMicroelectronics/stm32l0xx-hal-driver #4: Inconsistent CMSIS core tag in README of HAL vs. CMSIS device

dnmiller, 2021-05-18:
Hi,

In the README of this repo, it lists the necessary CMSIS core git tag as being v4.5_cm0, which is a valid tag in the CMSIS core repo, but in https://github.com/STMicroelectronics/cmsis_device_l0 the required tag is v5.4.0_cm0.

I suspect https://github.com/STMicroelectronics/cmsis_device_l0 has the correct tag listed in the docs. Whichever it is, please make them consistent. Thanks.

RKOUSTM, 2021-06-30:
Hi @dnmiller,

Thank you for your contribution. The issue you pointed out has been confirmed. An internal bug tracker has been logged and a fix will be implemented and made available in the frame of a future release. 

>- Concerning the issue related to the Readme.md of stm32l0xx_hal_driver, the proposed fix is as follows: 

HAL Driver L0 | CMSIS Device L0 | CMSIS Core      | Was delivered in the full MCU package
--------------- | -------------------| -----------------| --------------------------------------------------------
Tag v1.10.2     | Tag v1.9.0            | Tag v4.5.0_cm0 | Tag v1.11.2 (and following, if any, till next new tag)
Tag v1.10.3     | Tag v1.9.0            | Tag v4.5.0_cm0 | Tag v1.11.3 (and following, if any, till next new tag)
Tag v1.10.4     | Tag v1.9.1            | Tag v5.4.0_cm0 | Tag v1.12.0 (and following, if any, till next new tag)

>- For the Readme.md of cmsis_device_l0, a fix of the typo in the title of readme will be proposed. Also, an update of the table versions of the CMSIS Device component is proposed as follows:

CMSIS Device L0  | CMSIS Core       | Was delivered in the full MCU package
------------------  | ------------------| --------------------------------------------------------
Tag v1.9.0             | Tag v4.5.0_cm0  | Tag v1.11.0 (and following, if any, till next new tag)
Tag v1.9.1             | Tag v5.4.0_cm0  | Tag v1.12.0 (and following, if any, till next new tag)

Thank you once again for your contribution. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2021-06-30:
ST Internal Reference: 109918

RKOUSTM, 2021-08-13:
Hi @dnmiller,

The fix you requested has been implemented and is now available in the frame of the latest stm32l0_hal_driver package V1.10.5 release. This issue can be closed now. Thank you again for your contribution.

With regards,

closed 2021-08-13 by RKOUSTM