STMicroelectronics/stm32l0xx-hal-driver #6: HAL_ADC_Start_DMA() misleading param description

tdjastrzebski, 2021-12-29:
HAL_ADC_Start_DMA() method, the last `Length` param description reads:

> `* @param Length Number of data to be transferred from ADC peripheral to memory`

This is misleading. Param name and the description suggests data buffer length is required while number of conversions is expected.

This value is finally used to set `hdma->Instance->CNDTR` register (DMA channel x number of data to transfer register).
According to docs "It is decremented after each single DMA ‘read followed by write’ transfer, indicating the remaining **amount of data** items to transfer."

I think `HAL_DMA_Start_IT()` and `DMA_SetConfig()` methods suffer from a similar documentation ambiguity. `DataLength` param name is used while `DataLength` is not same as **amount of data**.

As far as I can tell, the problem exists in all HAL packages.

ASELSTM, 2022-01-31:
ST Internal Reference: 121956

ASELSTM, 2023-04-06:
Hi @tdjastrzebski,

Thank you for your contribution. This issue has been fixed in the frame of version v1.10.6 of the stm32l0xx_hal_driver. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM