STMicroelectronics/stm32l0xx-hal-driver #9: File stm32l0xx_hal_adc_ex.c.: Wrong information in function comments for HAL_ADCEx_EnableVREFINT() and HAL_ADCEx_EnableVREFINTTempSensor()

mosimasip, 2024-04-19:
File stm32l0xx_hal_adc_ex.c. Function comments for HAL_ADCEx_EnableVREFINT() and HAL_ADCEx_EnableVREFINTTempSensor() says "This API is obsolete. This configuration is done in HAL_ADC_ConfigChannel()". 

But this is wrong. 

stm32l0xx_hal_adc.c says they need to be called in some occasions and HAL_ADC_ConfigChannel() does not configure this.

RJMSTM, 2024-04-23:
Hello @mosimasip,
in stm32l0xx_hal_adc.c It is stated in the comment for HAL_ADC_ConfigChannel that
```
 In case of usage of internal measurement channels:
  *         VrefInt/Vlcd(STM32L0x3xx only)/TempSensor
```

[image]

If you use **VrefInt** or  **TempSensor**, you just need to respect the Sampling time.

Please can you clarify your request or give me more details about your problem?

Regards,
Rania

mosimasip, 2024-04-23:
As I states. The function comment for HAL_ADCEx_EnableVREFINT() and HAL_ADCEx_EnableVREFINTTempSensor() are wrong. They say "This API is obsolete. This configuration is done in HAL_ADC_ConfigChannel()".  They are not obsolete and HAL_ADC_ConfigChannel() does not configure the same things as HAL_ADCEx_EnableVREFINT() and HAL_ADCEx_EnableVREFINTTempSensor().

RJMSTM, 2024-04-23:
Hello @mosimasip,

The function HAL_ADC_ConfigChannel effectively configures VrefInt and the temperature sensor. Thus, it might be possible to do without the functions HAL_ADCEx_EnableVREFINT...(). However, the driver documentation states that these functions should be called upon exiting Low power mode. The exact reason is unclear, but it is likely to be valid.

In summary, these functions are only necessary when exiting LowPower mode. These APIs have been marked "obsolete" because their functionality has been integrated into the HAL_ADC_ConfigChannel function. And for LowPower mode, one could use LL_ADC_SetCommonPathInternalCh.

We will perform an update in the comment section of the `HAL_ADCEx_EnableVREFINT` , `HAL_ADC_ConfigChannelfunction` and `HAL_ADC_Init ` also in The introduction of .c

RJMSTM, 2024-05-06:
ST Internal Reference: 180731

RJMSTM, 2024-06-13:
Fixed in : 86984275fa7dbaa617713d0309575fc2c2fa1448

closed 2024-06-13 by RJMSTM