STMicroelectronics/stm32l0xx-hal-driver #10: Missing LIN break detection callback in UART HAL driver

NicoPy, 2024-06-06:
It is currently not possible to detect a line break on a UART using `stm32l0xx_hal_uart.h` and `stm32l0xx_hal_uart.c`.
This has been requested long ago [here](https://community.st.com/t5/stm32-mcus-products/stm32f09xx-usart-lin-break-interrupt-management-using-hal/m-p/306369) but it has still not been implemented. Any reason for that ?

KRASTM, 2024-06-10:
ST Internal Reference: 183583

KRASTM, 2024-06-11:
Hello @NicoPy,

Regarding your question about Auto baud rate detection, unfortunately there is no intention to add this feature to this series for the moment, may be in the future, I will keep you informed. 
However, there is an example in H5 series `STM32CubeH5\Projects\NUCLEO-H533RE\Examples\UART\UART_AutoBaudrate_Detection`, you can use it, to adapt and use this feature on L0.

With regards,

NicoPy, 2024-06-11:
Hello @KRASTM ,
Thanks for your answer. 
Today I realized there is no interrupt for Auto Baud Rate. Just flags indicating finished with/without error when a character has been received.
So, a callback is not needed. 
Regards,
Nicolas

KRASTM, 2025-01-20:
Hello @NicoPy,

I hope you're doing well.

Regarding the issue, our team has decided that there is no current plan to address it. If there are any updates, I will keep you informed. Thank you in advance for your understanding.

Please, allow me to close the issue.

With regards.

closed 2025-01-20 by KRASTM

NicoPy, 2025-01-20:
Hello,

Thank you for letting me know.

Regards.