STMicroelectronics/stm32l0xx-hal-driver #11: The macro __HAL_FLASH_CLEAR_FLAG does not work properly

tomatsu, 2024-06-17:
I am experiencing an error when I write data to EEPROM.  I suspect that __HAL_FLASH_CLEAR_FLAG does not property clear FLASH->SR, and the macro should be defined as follows.

`#define __HAL_FLASH_CLEAR_FLAG(__FLAG__)  ((FLASH->SR) &= ~(__FLAG__))`

I would appreciate if you could take a look at the code.

TOUNSTM, 2024-06-20:
Hello @tomatsu,

Thank you for your contribution, In fact, these flags are cleared by writinig 1 according to [RM0394 Rev 4](https://www.st.com/resource/en/reference_manual/rm0394-stm32l41xxx42xxx43xxx44xxx45xxx46xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) page 102. Here's how you might define a macro to clear a flag by writing a 1:
`#define __HAL_FLASH_CLEAR_FLAG(__FLAG__)  ((FLASH->SR) = (__FLAG__)) ` which is correct in our driver. 
Regarding the point for which you opened the issue, as it is irrelevant, allow me to close the issue.

TOUNSTM, 2024-06-20:
Hello @tomatsu,

Thank you for your contribution, the macro you provided:
`#define __HAL_FLASH_CLEAR_FLAG(__FLAG__)  ((FLASH->SR) &= ~(__FLAG__))` 

 is intended to clear specific flags by performing a bitwise AND operation with the complement of the flag. This method is typically used for flags that are **reset by writing 0**. However, as per [RM0394 Rev 4](https://www.st.com/resource/en/reference_manual/rm0394-stm32l41xxx42xxx43xxx44xxx45xxx46xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) page 102 the flags in the `FLASH->SR `register are **cleared by writing 1**. 

Therefore, the correct definition should be:
`#define __HAL_FLASH_CLEAR_FLAG(__FLAG__)  ((FLASH->SR) = (__FLAG__)) ` 

This aligns with our driver's implementation. As for the issue you raised, since it is not relevant, I will proceed to close it. Thank you for your comprehension.

With regards,

closed 2024-06-20 by TOUNSTM