STMicroelectronics/stm32l0xx-hal-driver #12: HAL_RCC_GetSysClockFreq() uses 64 bit arithmetic when 32 would suffice

kasjer, 2024-06-18:
Function HAL_RCC_GetSysClockFreq() casts numbers to uint64_t when calculating pllvco.
This is far beyond what is needed and results in inclusion of additional code from gcc library and in turn increases code size.
In my case code size was increased for over 700 bytes for no reason.

**Describe the set-up**
 * nucleo-l073rz
 * arm-none-eabi-gcc 10.3

**Additional context**
Problematic lines:
https://github.com/STMicroelectronics/stm32l0xx_hal_driver/blob/86984275fa7dbaa617713d0309575fc2c2fa1448/Src/stm32l0xx_hal_rcc.c#L1245
https://github.com/STMicroelectronics/stm32l0xx_hal_driver/blob/86984275fa7dbaa617713d0309575fc2c2fa1448/Src/stm32l0xx_hal_rcc.c#L1251
https://github.com/STMicroelectronics/stm32l0xx_hal_driver/blob/86984275fa7dbaa617713d0309575fc2c2fa1448/Src/stm32l0xx_hal_rcc.c#L1255

Removing casts reduces code size without any additional penalty.

TOUNSTM, 2024-06-21:
ST Internal Reference: 184777

KRASTM, 2024-09-18:
Fixed in commit https://github.com/STMicroelectronics/stm32l0xx_hal_driver/commit/3611fb0b6c3cb087f222f17a65cb8c5489459b21

closed 2024-09-18 by KRASTM